jasonacox/Powerwall-Dashboard #647: removing my recommendation for the TP-Link AC1300 Dual Band WiFi Adapter -Archer T3U plus

JonMurphy, 2025-06-26:
Jason - I am removing my recommendation for the TP-Link AC1300 Dual Band WiFi Adapter -Archer T3U plus (RTL8812BU 2357:0138) at <https://www.amazon.com/TP-Link-usb-wifi-adapter-pc/dp/B08D72GSMS?th=1>.  See https://github.com/jasonacox/Powerwall-Dashboard/issues/621#issuecomment-2868406348

All worked a-ok on a RPi5 from the beginning of May until near June 17,18,19.  My wi-fi connection keeps dropping and data to the P-D stops.

The best I can tell sudden disconnects are a feature when using the RTL8812BU 2357:0138.

```
Jun 26 08:34:12 teslastuff NetworkManager[705]: <info>  [1750944852.8674] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Jun 26 08:34:31 teslastuff NetworkManager[705]: <warn>  [1750944871.1800] device (wlan0): Activation: (wifi) association took too long
Jun 26 08:34:31 teslastuff NetworkManager[705]: <info>  [1750944871.1801] device (wlan0): state change: config -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Jun 26 08:34:31 teslastuff NetworkManager[705]: <warn>  [1750944871.1806] device (wlan0): Activation: failed for connection 'TeslaPW_REDACT'
Jun 26 08:34:31 teslastuff NetworkManager[705]: <info>  [1750944871.1808] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Jun 26 08:35:02 teslastuff NetworkManager[705]: <info>  [1750944902.0574] manager: (veth6b24f5b): new Veth device (/org/freedesktop/NetworkManager/Devices/65)

Jun 26 08:56:53 teslastuff NetworkManager[705]: <info>  [1750946213.1886] device (wlan0): disconnecting for new activation request.
Jun 26 08:56:53 teslastuff NetworkManager[705]: <info>  [1750946213.1887] audit: op="connection-activate" uuid="88478eac-REDACT-98a4035cbf6d" name="TeslaPW_REDACT" pid=218176 uid=0 result="success"
Jun 26 08:56:55 teslastuff NetworkManager[705]: <warn>  [1750946215.3603] device (wlan0): Deactivation failed: GDBus.Error:fi.w1.wpa_supplicant1.NotConnected: This interface is not connected
Jun 26 08:56:55 teslastuff NetworkManager[705]: <info>  [1750946215.3604] device (wlan0): state change: deactivating -> disconnected (reason 'new-activation', sys-iface-state: 'managed')
Jun 26 08:56:55 teslastuff NetworkManager[705]: <info>  [1750946215.3614] device (wlan0): Activation: starting connection 'TeslaPW_REDACT' (88478eac-REDACT-98a4035cbf6d)
Jun 26 08:56:55 teslastuff NetworkManager[705]: <info>  [1750946215.3621] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
```

**wlan0 is the TP-Link WiFi Adapter**

With the above errors I can _sometimes_ get the wi-fi link back by entering:

```
sudo nmcli device connect wlan0

# or sometimes

sudo nmcli d wifi connect "TeslaPW_REDACT" password "REDACT" ifname wlan0
```

Today nothing seems to help!

I think the DROPs started after an `apt dist-upgrade`:

```
--
Start-Date: 2025-06-17  14:39:45
Commandline: apt dist-upgrade
Requested-By: pi (1000)
Upgrade: rpi-eeprom:arm64 (28.1-1, 28.2-1), libpipewire-0.3-common:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), pipewire-pulse:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), libspa-0.2-libcamera:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), raspi-config:arm64 (20250605, 20250612), pipewire:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), wf-panel-pi:arm64 (0.101, 0.102), python3-pidng:arm64 (4.0.9+git20250121.e2f5a8d-1, 4.0.9+git20250603.8427835-1), firefox:arm64 (139.0-1+rpt1, 139.0.4-1+rpt1), pipewire-libcamera:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), libgstreamer-plugins-bad1.0-0:arm64 (1.22.0-4+deb12u5, 1.22.0-4+deb12u6), libspa-0.2-bluetooth:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), pipewire-bin:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), ca-certificates:arm64 (20230311, 20230311+deb12u1), libspa-0.2-modules:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), gstreamer1.0-plugins-bad:arm64 (1.22.0-4+deb12u5, 1.22.0-4+deb12u6), libpipewire-0.3-modules:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5), libpipewire-0.3-0:arm64 (1.2.7-1~bpo12+1+rpt3, 1.2.7-1~bpo12+1+rpt5)
End-Date: 2025-06-17  14:39:55
--

--
Start-Date: 2025-06-22  16:41:20
Commandline: apt dist-upgrade
Requested-By: pi (1000)
Upgrade: libgdk-pixbuf2.0-bin:arm64 (2.42.10+dfsg-1+deb12u1, 2.42.10+dfsg-1+deb12u2), libblockdev-part2:arm64 (2.28-2, 2.28-2+deb12u1), libblockdev-swap2:arm64 (2.28-2, 2.28-2+deb12u1), gir1.2-gdkpixbuf-2.0:arm64 (2.42.10+dfsg-1+deb12u1, 2.42.10+dfsg-1+deb12u2), libblockdev-fs2:arm64 (2.28-2, 2.28-2+deb12u1), udisks2:arm64 (2.9.4-4, 2.9.4-4+deb12u1), libgdk-pixbuf-2.0-0:arm64 (2.42.10+dfsg-1+deb12u1, 2.42.10+dfsg-1+deb12u2), libudisks2-0:arm64 (2.9.4-4, 2.9.4-4+deb12u1), libc6:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), locales:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libblockdev-loop2:arm64 (2.28-2, 2.28-2+deb12u1), libblockdev2:arm64 (2.28-2, 2.28-2+deb12u1), libblockdev-part-err2:arm64 (2.28-2, 2.28-2+deb12u1), libblockdev-utils2:arm64 (2.28-2, 2.28-2+deb12u1), libc-dev-bin:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libc-l10n:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libc-bin:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libc-devtools:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libc6-dbg:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libc6-dev:arm64 (2.36-9+rpt2+deb12u10, 2.36-9+rpt2+deb12u12), libblockdev-crypto2:arm64 (2.28-2, 2.28-2+deb12u1), libgdk-pixbuf2.0-common:arm64 (2.42.10+dfsg-1+deb12u1, 2.42.10+dfsg-1+deb12u2)
End-Date: 2025-06-22  16:41:29
--

jasonacox, 2025-06-27:
@JonMurphy are you switching to something else?  I've had good luck with:

* Panda Wireless® PAU0D AC1200 USB Adapter w/Dual Antennas -  https://www.amazon.com/dp/B0B2QD6RPX?ref=ppx_yo2ov_dt_b_fed_asin_title 

I would love to start a list of WiFi adapters that are performing well.

JonMurphy, 2025-06-27:
Yes, getting a new one.  I ordered the Panda Wireless PAU0D AC1200 USB Adapter.  I believe it has a MediaTek MT7612U chipset and those are suppose to be linux friendly.  **No more Realtek wi-fi adapters for me!**

I also order a EDUP USB WiFi Adapter with High Gain 6dBi Antenna Support (with MediaTek MTK 7601U chipset) - <https://www.amazon.com/dp/B0827LG8L2>.  It is for 2.4GHz only.  

I'll try the EDUP first so maybe we have three items in the list.

| wi-fi adapter                                          | chipset                    | recommend |
| ------------------------------------------------------ | -------------------------- | --------- |
| Panda Wireless® PAU0D AC1200 USB Adapter               | MediaTek MT7612U           | yes       |
| TP-Link AC1300 Dual Band WiFi Adapter -Archer T3U plus | RealTek RTL8812BU 2357:013 | never!    |
| EDUP USB WiFi Adapter with High Gain 6dBi Antenna      | MediaTek MTK 7601U         | TBD       |

jasonacox, 2025-06-28:
Thanks @JonMurphy -  let us know how it goes for you!