jasonacox/Powerwall-Dashboard #655: Power Flow Animation not working on Grafana v11+

s-crypt, 2025-07-07:
**Problem**
When Grafana is v10+, the Power Flow visualization does not work

**To Reproduce**
1. Upgrade to Grafana v10+ in powerwall.yml
2. View Power Flow panel not working

**Screenshots**
[image]

**System Details**
 - Powerwall Installation: PW2
 - Powerwall Firmware Version: 25.18.2
 - Host System Hardware and OS: RasPi 3B+, Raspbian (latest)

**Additional context**
Powerwall Dashboard Installation: Local

Hopefully this isn't out of scope, but the power flow animation does not seem to work in the latest versions of grafana. I am a stickler for having everything up-to-date for security reasons, but recently the panel has stopped working in the past month or so.

The firmware version is able to be successfully retrieved for the panel, and the image blobs for the animation are loaded (seen w/ browser tools), but the panel is just blank. On the stock version 9.1.2 the panel loads fine. The animation is working when visiting the pypowerwall URL directly (:8675).

I am terrible at HTML/CSS/JS so some assistance in getting the panel's code working would be greatly appreciated!

[My current dashboard](https://gist.github.com/s-crypt/41f66ac5ee902b2e2c6db512e3aeb1c3) (working on Grafana 10+)

youzer-name, 2025-07-07:
I think you typo'd the Grafana version in your probelm statement and 'to reproduce' steps.  I assume you mean it was working on Grafana 10.x and not working on Grafana 11.x

See my prior post on this issue here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/611

s-crypt, 2025-07-08:
@youzer-name sorry, you're right

s-crypt, 2025-07-08:
@youzer-name you are awesome! the height change in the css did the trick!

closed 2025-07-08 by s-crypt