jasonacox/Powerwall-Dashboard #670: After updating from 4.8.1 to 4.8.3 pypowerwall fails to start with a access denied issue

Meowcat285, 2025-07-20:
After updating to 4.8.3 from 4.8.1 I now get a access denied error from pypowerwall. I am using the TEDAPI method and am using a RPi as a relay to the powerwall. When I downgrade back to 4.8.1 by restoring a backup, pypowerwall is suddenly able to connect just fine. Below is the log output from pypowerwall after upgrading to 4.8.3

```
07/20/2025 12:43:52 PM [proxy] [INFO] pyPowerwall [0.13.2] Proxy Server [t78] - HTTP Port 8675
07/20/2025 12:43:52 PM [proxy] [INFO] pyPowerwall Proxy Started
07/20/2025 12:43:52 PM [proxy] [INFO] Network error rate limiting: 5 errors/minute per function
07/20/2025 12:43:52 PM [proxy] [INFO] Graceful degradation enabled (PW_GRACEFUL_DEGRADATION=yes) - cached data TTL: 30s
07/20/2025 12:43:52 PM [proxy] [INFO] Connection health monitoring enabled (PW_HEALTH_CHECK=yes)
07/20/2025 12:43:52 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
07/20/2025 12:43:52 PM [pypowerwall.tedapi] [ERROR] Failed to connect to Powerwall Gateway
07/20/2025 12:43:53 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
07/20/2025 12:43:53 PM [pypowerwall.fleetapi.pypowerwall_fleetapi] [ERROR] Failed to retrieve sitelist - MissingSchema("Invalid URL '/api/1/products': No scheme supplied. Perhaps you meant https:///api/1/products?")
07/20/2025 12:43:53 PM [pypowerwall.fleetapi.pypowerwall_fleetapi] [ERROR] No sites found for 
07/20/2025 12:43:53 PM [pypowerwall] [ERROR] Failed to connect to Powerwall with all modes. Waiting 30s to retry.
```

jasonacox, 2025-07-21:
It looks like it attempted to connect to the TEDAPI address (192.168.91.1) and failed and then proceeded to attempt FleetAPI. Can you provide the pypowerwall details you currently see with `verify.sh`?  I wonder if you are using local mode. Also, to test the new code, can you try this (will not change your current install):

```bash
pip install pypowerwall -U
python -m pypowerwall.tedapi
```

In your pypowerwall.env file make sure it lists the correct password for `PW_GW_PWD`.

Meowcat285, 2025-07-21:
> It looks like it attempted to connect to the TEDAPI address (192.168.91.1) and failed and then proceeded to attempt FleetAPI. Can you provide the pypowerwall details you currently see with `verify.sh`? I wonder if you are using local mode. Also, to test the new code, can you try this (will not change your current install):
> 
> pip install pypowerwall -U
> python -m pypowerwall.tedapi
> 
> In your pypowerwall.env file make sure it lists the correct password for `PW_GW_PWD`.

Here's the out put from `verify.sh` pre-upgrade

```
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.13.1 Proxy t75
 - Powerwall State: CONNECTED - Firmware: 25.18.4
 - Site Name: My Home
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: hybrid
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 92.39}
 - Version: 0.2.3

All tests succeeded.
```

And here's the output after the upgrade
```
Verify Powerwall-Dashboard 4.8.3 on Linux - Timezone: America/New_York
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Tip: If colors are hard to read, try: ./verify.sh --no-color, --lightbg, or --darkbg
[... 151 lines cut ...]
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 92.97}
 - Version: 0.2.3

One or more tests failed.
```

So I'm a bit stumped on what's going on here

Meowcat285, 2025-07-22:
> You mentioned using a RPi but are you using routing or directly connected?
The RPi is connected to the powerwall directly over WiFi and to the rest of my network over ethernet, and I have a routing rule on the server to route traffic to 192.168.91.1 via the RPi.

Meowcat285, 2025-07-22:
The other thing to note is that the TEDAPI works fine on 4.8.1, with all enhanced metrics showing up, but when I update, that's when I start having that access denied issue

jasonacox, 2025-07-26:
@Meowcat285 thank you! Do you have another system where you could try to install from scratch to see if it behaves differently? I simply can't reproduce anything similar so I'm wondering what is so different about your installation. 

Has anyone else had a problem with the upgrade? Please chime in.

Meowcat285, 2025-07-26:
> [@Meowcat285](https://github.com/Meowcat285) thank you! Do you have another system where you could try to install from scratch to see if it behaves differently? I simply can't reproduce anything [...]

I'll give that a try, I'll let you know if it happens there as well

jasonacox, 2025-07-27:
I don't think there is anything in the new pypowerwall that would change your experience, but I have updated to v4.8.4 with the new pypowerwall v0.14.0. Focus of that update is more about Cloud and FleetAPI access vs TEDAPI local access.

Meowcat285, 2025-07-30:
Sorry for taking so long to get around to this (I got a bit busy). I created a new VM to try it, and I'm still getting that error in a fresh VM.

> I don't think there is anything in the new pypowerwall that would change your experience, but I have updated to v4.8.4 with the new pypowerwall v0.14.0. Focus of that update is more about Cloud and FleetAPI access vs TEDAPI local access.

Nope, doesn't fix it when upgrading the old VM or when installing the new one

jasonacox, 2025-07-31:
Ok, trying to think through things... What changed?

* https://github.com/jasonacox/Powerwall-Dashboard/compare/v4.8.1...v4.8.3 -> Mainly pypowerwall version from jasonacox/pypowerwall:0.13.1t75 to jasonacox/pypowerwall:0.13.2t78 
* v0.13.1 to v0.13.2 (https://github.com/jasonacox/pypowerwall/compare/v0.13.1...v0.13.2) -> Mainly adds cache logic for calls to reduce load on TEDAPI and better handle backoff/cooldown errors.

I can figure what could be in that setup that is only affecting your installation. I appreciate your help with this, would you mind testing a few more things to help gather diagnostics on the new VM that doesn't work?

* Do you have the Tesla One mobile app? If so, you should be able to use that to connect to the WiFI and log in to the Powerwall. That will help us confirm there isn't an issue with the password. I'm pretty sure that it requires the full password for the PW3. Can you try full password again? Also, please update `PW_GW_PWD` and not `PW_PASSWORD` if you are editing pypowerwall.env directly.
* Run pypowerwall in debug mode - change the pypowerwall.env to have `PW_DEBUG=yes` and restart `./compose-dash.sh up -d` and look at logs `docker logs pypowerwall`.
* Try using v4.8.1 on your new VM to see if it works: `git checkout tags/v4.8.1` and rerun setup.sh to see if the older version works. 

> /app $ python -m pypowerwall.tedapi

I just noticed "/app" in this. Are you running this as a docker process (e.g. docker exec -it pypowerwall sh)? If so can you install pypowerwall natively in your VM to run the test? I think the error you saw was related to pypowerall proxy not being able to write to the local /app directory.

```bash
pip install pypowerwall==0.13.2 # or pip install pypowerwall==0.13.1 to test older one
python -m pypowerwall.tedapi
```

I would love to solve this puzzle.

Meowcat285, 2025-08-06:
UPDATE: I changed nothing and now 4.8.1 is now working on the new VM after restarting the compose stack a few times.

And now suddenly 4.8.4 is working now too. I'm gonna try updating the old VM and see if that works now

Meowcat285, 2025-08-06:
Update 2: And now updating to 4.8.4 works fine on the old VM.

I'm starting to suspect that something wonky was going on with my PW, and they recently received a firmware update, so it could be either that the restart triggered by the firmware update, or the firmware update itself fixed it.

I'm gonna leave this issue open for a few more days at most, just in case something breaks again

jasonacox, 2025-08-10:
That's fascinating! I wish we knew what happened here. Another user resported having to do a full hour long power cycle to get their Powerwalls working again (https://github.com/jasonacox/Powerwall-Dashboard/issues/680#issuecomment-3166172493). I guess it could be firmware related but seems odd that it only impacts some systems and not others.

jwvanderbeck, 2025-08-20:
Sadly I'm having this same issue after upgrading.  I really don't want to power cycle the powerwalls.  That cuts power to the house yeah?

jwvanderbeck, 2025-08-20:
> python -m pypowerwall.tedapi

Gives an access denied as well, regardless of whether I use the full password (which is what I was using prior to the update) or the last 5.  Could my password have been changed somehow over the air without my knowledge?

jasonacox, 2025-08-21:
Can you try `python -m pypowerwall.tedapi --debug` just to see if there is a clue as to cause?

Also, as mentioned above, try the older version that was working for you to see if it helps:

```bash
# revert to older pypowerwall
pip install pypowerwall==0.13.2 # or pip install pypowerwall==0.13.1 to test older one

python -m pypowerwall.tedapi
```

jwvanderbeck, 2025-08-21:
Current version debug:

```python -m pypowerwall.tedapi --debug
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.14.1]
DEBUG:pyPowerwall TEDAPI version 0.14.1
DEBUG:Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
DEBUG:Checking Powerwall Gateway at https://192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 404 19
DEBUG:Connection to Powerwall Gateway successful, code 404.
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: ********

Connecting to Powerwall Gateway 192.168.91.1
DEBUG:Testing Connection to Powerwall Gateway: 192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 404 19
DEBUG:Detected Powerwall 3 Gateway
DEBUG:Fetching DIN from Powerwall...
DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114
ERROR:Access Denied: Check your Gateway Password
ERROR:Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again```

The older versions give the same result.  I'm very confused because this was working fine until recently and I didn't change the passwords at all, and they are still there in the `pypowerwall.env` file.

jwvanderbeck, 2025-08-22:
Problem was the old route, though it for some reason didn't cause issues on the first instance.  In any case once I deleted that route it started working again on the second instance.  So all good now thanks!