jasonacox/Powerwall-Dashboard #679: I blew-up my P-D install when trying 3-FleetAPI to 1-Local

JonMurphy, 2025-08-06:
I broke my P-D!

I setup a spare RPi3 with **com.tesla.3p.public-key.pem** and I was giving **3 - FleetAPI Cloud** a try when things went bad.   I was in this area of `./setup.sh`:

```
Tesla FleetAPI Setup
--------------------

Step 1 - Register your application at https://developer.tesla.com/
Step 2 - Run create_pem_key.py to create a PEM key file for your website.
         Put the public key in {site}/.well-known/appspecific/com.tesla.3p.public-key.pem
Step 3 - Run this script to generate a partner token, register your partner account,
         generate a user token, and get the site_id and live data for your Tesla Powerwall.

No configuration found - Creating: .auth/.pypowerwall.fleetapi
```

I got this error:
```
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='https', port=443): Max retries exceeded with url: /redeacted/.well-known/appspecific/com.tesla.3p.public-key.pem (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7fff5b753010>: Failed to resolve 'https' ([Errno -2] Name does not resolve)"))
```
<p>

I figured out I didn't open the firewall properly.  My bad!

---

I fixed the firewall and I tried the **3-FleetAPI Cloud** again and I noticed this oddity:

 - setup thought I was using **2-Tesla Cloud** in the of past
 - I was not.  I was using **1-Local Access**.

```
pi@teslastuff:~/Powerwall-Dashboard $ ./setup.sh
Powerwall Dashboard (v4.8.4) - SETUP
-----------------------------------------
Select configuration mode:

Current: Tesla Cloud

 1 - Local Access     (Powerwall 1, 2, or + using the Tesla Gateway on LAN) - Default
 2 - Tesla Cloud      (Solar-only systems or Powerwalls without LAN access)
 3 - FleetAPI Cloud   (Powerwall systems using Official Telsa API)
 4 - Extended Metrics (Powerwall 2, +, or 3 using TEDAPI and local WiFi access)

Select mode: [2] 3

You are already using the Tesla Cloud configuration, are you sure you wish to change? [y/N] y

Timezone (leave blank for America/Chicago)
```

As I continued, I see this error:
```
Running Docker Compose...
[+] Running 5/5
 ✔ Container pypowerwall  Running                                                              0.0s
 ✔ Container influxdb     Running                                                              0.0s
 ✔ Container weather411   Running                                                              0.0s
 ✔ Container grafana      Running                                                              0.0s
 ✔ Container telegraf     Running                                                              0.0s
-----------------------------------------
pyPowerwall [0.14.0] - FleetAPI Mode Setup

Unable to refresh token. Response code: 400
Token expired, refresh token failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.10/runpy.py", line 86, in _run_code
[... 86 lines cut ...]
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 125
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 111
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 245
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 115
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 118
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 104
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 104
2025/08/06 19:06:06 error: error parsing query: found /, expected ON at line 1, char 41
2025/08/06 19:06:06 error: error parsing query: found /, expected ON at line 1, char 41
2025/08/06 19:06:06 error: error parsing query: found /, expected ON at line 1, char 41
2025/08/06 19:06:06 error: error parsing query: found ), expected GROUP BY time(...), time dimension must have duration argument at line 1, char 220
2025/08/06 19:06:06 error: error parsing query: found /, expected ON at line 1, char 38
2025/08/06 19:06:06 error: error parsing query: found /, expected ON at line 1, char 38
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 120
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 326
2025/08/06 19:06:06 error: error parsing query: found ), expected GROUP BY time(...), time dimension must have duration argument at line 1, char 253
2025/08/06 19:06:06 error: error parsing query: found ), expected GROUP BY time(...), time dimension must have duration argument at line 1, char 345
2025/08/06 19:06:06 error: error parsing query: found year, expected GROUP BY time(...), time dimension must have duration argument at line 1, char 191
2025/08/06 19:06:06 error: error parsing query: found /, expected INTO at line 1, char 104
Fetching local weather...
weather411
------------------[ Final Setup Instructions ]-----------------
```

Help?!?  What did I screw up??

JonMurphy, 2025-08-06:
I think I messed up the database since the graphs stopped:

[image]

JonMurphy, 2025-08-07:
I think the database is ok.  I can run [Tesla history](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history) and all of the old data appears in the P-D graphs.  

But no new data moves from the Powerwall to P-D (except the **Power Flow** graphic on right.)

I've tried compose down/up.  I've tried a [Restore](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/backups/README.md#restore-backup).  And I tried another ./setup.sh.  Still no new data.

help?!?

SDWolf, 2025-08-07:
I also lost any new data showing up, other than charge %.  It stopped early this morning without any changes being made.  The power flow will start and stop randomly, but mostly only showing battery charge.  

[image]

I have tried a reset of my PW3s.  The verify services shows all tests succeeded:

Verify Powerwall-Dashboard 4.8.4 on Linux - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Tip: If colors are hard to read, try: ./verify.sh --no-color, --lightbg, or --darkbg

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.14.0 Proxy t80
 - Powerwall State: CONNECTED - Firmware: 25.26.0
 - Site Name: Skynet
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: full 
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 74.66}
 - Version: 0.2.3

All tests succeeded.

JonMurphy, 2025-08-07:
Got it.  The `./Powerwall-Dashboard/telegraf.conf` file was corrupt.  

In the image below the left file is an old `telegraf.conf` backup file and the right file was my corrupt `telegraf.conf` file.  Once I replaced the `telegraf.conf` file all started working again!

As a VERY wild guess, I think this happened when I ran `./setup.sh` and entered a `1` as the answer to `Enter Timezone: `.  Again this is a wild guess!

[image]

JonMurphy, 2025-08-07:
@SDWolf - the issue posted is not related to this one.  Please delete the above post:

[image]

Then go to the top of a page and click the green **New Issue** and I will answer.

PR #681 (2025-08-10): v4.8.5
This pull request introduces improvements to setup and verification scripts, focusing on enhanced log display options and robust input validation for timezones. These changes help prevent configuration errors and provide a cleaner, more user-friendly setup experience.

**Setup and Verification Enhancements:**

* Added enhanced log display options to `verify.sh`, including support for `--logs` and `--no-logs` flags, and an interactive prompt for log viewing. Log output is now cleaner and only shown when requested.
* Implemented input validation for timezone entry in `setup.sh`, ensuring  [... cut]
closed 2025-08-10 by jasonacox